baruch/b3603 #4: Alert when the unit is fed below 6V

baruch, 2015-02-17:
Accurate will suffer considerably if the unit is fed less than 6V and it gets pretty inaccurate once you go below 5V. The device should alert about this, it should allow operation but give a warning to let the user know he is outside bounds.

Paul-Ver, 2016-05-04:
I'm a little late to the party, but this could be done by checking the voltage on Pin 20?
And if it's lower as 6V, write a specific message to the screen?

I'm thinking of buying the module as supply, but modding it would be a fun way to get started with ST microcontrollers.

baruch, 2016-05-04:
I really can't recall much details these days, it's been a long while since I worked on this. You are welcome to buy and try it out as I'm busy with other things to be able to return to this topic for now (no immediate use for now).

I would recommend buying two of these, they are rather sensitive and open frame so any short you mistakenly create will damage something.

It is fun though! :-)