baruch/b3603 #11: Use the buttons

baruch, 2015-02-24:
Find how to make use of the buttons and integrate them into a UI that doesn't necessarily depend on the serial port.

UsrnameTaken, 2017-09-03:
Please [take a look](https://github.com/UsrnameTaken/B3606/tree/master/docs/b3606_uswitch) at screenshots from oscilloscope of pressing those two strange connected buttons (OK and UP).

iafilius, 2017-11-27:
Hi Baruch and others

I like to share the way the buttons work, remained a secret for 2 years now.

Howto:
Default ports pd1 and pc7 are set as input and can read (primary) buttons set and down directly and are pushed when eq zero)

If no primary key is pressed, seconday keys can be scanned (up and ok buttons).

Scan button 4 (ok):
Pd1 output value set to zero
Pd1 define as output (set direction)
Read pc7, key pressed when zero
Return pd1 to input in pull-up mode

Scan button 3 (up)
Then the other way around
Pc7 output value to zero.
Pc7 define as output
Read pd1, key pressed when zero
Return pc7 to input with pull-up.

I had worries on triggering the SWIM logic on pd1, but until now i did not show any reactions.
You may disable it by setting in the cdg_GCR bit0 to 1

When scanning secondary keys the first line returned to pull-up needs quite some recovery time. About 10 NOP’s makes it ready for next button scan.

Wrote some test code and was able to read set,down,up and ok key reliable.

Currently called from main event loop as demo, maybe better call it from a timer with handler.

While make the implementation i had to remove random serial stuff else it didn’t flash as firmware was too big.

I can provide working demo code and  scope images of key presses.
Those already provided 

Regards, Arjan

baruch, 2017-11-28:
@iafilius thanks!

I haven't worked on the b3603 for a while, I'll try to see if I can integrate it into the firmware. It was a pain to use the device without the buttons.

As for the firmware size, there is another fork of this project that worked on reducing the size, I'll probably try to integrate that as well to reduce the size.

iafilius, 2017-11-28:
Hi, uploaded the tested code , easy to integrate in main event loop (or called from timer)
uploaded the key scope pictures as well.
[buttons.c.txt](https://github.com/baruch/b3603/files/1510990/buttons.c.txt)
[buttons.h.txt](https://github.com/baruch/b3603/files/1510991/buttons.h.txt)
[image]
[image]
[image]
[image]
[image]