baruch/b3603 #14: Code is too large to fit STM8

UsrnameTaken, 2017-08-02:
After running make I get this:
```
CC main.c
CC display.c
CC uart.c
CC eeprom.c
CC outputs.c
CC config.c
CC fixedpoint.c
CC parse.c
CC adc.c
LINK b3603.ihx
Code is too large, it is 8570 bytes
Makefile:12: commands for 'check_size' object failed
make: *** [check_size] error 1
```

baruch, 2017-08-02:
What version of sdcc are you using?
please show the output of ```sdcc -v```

Mine is 3.4.3 and the code generated is 7915 bytes.

baruch, 2017-08-02:
You can also use the command ```make SHELL='sh -x'``` to show the commands and make sure that you have the option ```--opt-code-size``` in there.

UsrnameTaken, 2017-08-03:
I am using v. 3.4.0 since newer version had some strange problem.
`SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.4.0 #8981 (Apr  5 2014) (Linux)`

The code was downloaded and straight from this I tried to compile. Makefile has `--opt-code-size` in `CFLAGS=` but `make SHELL='sh -x'` shows:
`+ grep ^CODE b3603.map+ head -n1

+ sed -e s/^.*=\s*\([0-9]\+\).*$/\1/
+ CODESIZE=8570
+ [ 8570 -gt 8192 ]
+ echo Code is too large, it is 8570 bytes > 8192
Code is too large, it is 8570 bytes > 8192
+ exit 1
Makefile:12: commands for 'check_size' object failed
make: *** [check_size] error 1`

baruch, 2017-08-03:
To see the command you need to first make clean and then rebuild with the
shell option.

baruch, 2017-08-03:
There is no real reason for the size difference other than the compiler. There are likely missing optimizations in your version and not much I can do about it directly.

If you want you can try a fork of this project that also tried to reduce size from https://github.com/swegener/b3603/tree/size-reduction

UsrnameTaken, 2017-08-03:
Thanks to you problem is solved. I couldn't find v. 3.4.3, but 3.5.0 works good with additional option.
I've tried compiling reduced size version, but with 3.4.0 the result was worse, than regular.

closed 2017-08-04 by baruch