baruch/b3603 #15: Board not responding to serial commands

UsrnameTaken, 2017-08-17:
I've programmed a fresh STM8S with provided code, but it looks like there is no response from board when it starts or when I write to it through serial port. CC (yellow) LED is always on.
I've tested my USB-SERIAL adapter and Intel Galileo serial port. Both working.
STM was programmed using stm8flash/STVP and ST-LINK/V2 mini.

baruch, 2017-08-19:
I understand you are working on the B3606. Are you sure the stm8 is the same as the one I have on the b3603?

When you just connect the serial and turn on the input power to the board you should see data on the serial port. It should be configured to 38400/8n1

UsrnameTaken, 2017-08-19:
Yes, my board is B3606. The only differences I found are in the converter part, the rest (top board and bottom board under top board) is the same.
I was working on 9600/8n1 as mentioned in PROTOCOL.md. There was no welcome message or response to query. I will try different baud rates tommorow in the evening.

UsrnameTaken, 2017-08-21:
Ok, the board is deaf. Powering it from Galileo - nothing, powering from external ATX PS - nothing, changed TX with RX multiple times - nothing, looked into code for UART speed (indeed 38400) - nothing, welcome message - not present, GND not/connected to Galileo - nothing.
I can be sure Galileo is working properly, shorting Rx with Tx gives me MMOODDEELL when writing MODEL. I have tested USB to serial adapter also.
I don't think so, but maybe my changes in voltage/current limits broke the program? I should try programming the STM with original source.

UsrnameTaken, 2017-08-22:
Ok, the board is deaf. Powering it from Galileo - nothing, powering from external ATX PS - nothing, changed TX with RX multiple times - nothing, looked into code for UART speed (indeed 38400) - nothing, welcome message - not present, GND not/connected to Galileo - nothing.
I can be sure Galileo is working properly, shorting Rx with Tx gives me MMOODDEELL when writing MODEL. I have tested USB to serial adapter also.
I don't think so, but maybe my changes in voltage/current limits broke the program? I should try programming the STM with original source.

[Addition]
After checking the STM for short circuits I have found two points of unnecessary, but invisible connection - pin 4 and 5 (NRST + PA1), pin 9 and 10 (VDD + (HS)_PA3). I will remove them and try again.

UsrnameTaken, 2017-08-25:
Removing short circuits and programming again helped a lot :)
For now the board says: B3606 starting Version...
After this it starts to send "PC3 is now 0" and "PC3 is now 1" alternately without any control. The screen fills with this oblique pattern. It's from line 489 in main.c function read_state. As I see in the schematic of top board PC3 is unconnected. Is this intended?

baruch, 2017-08-25:
I believe this was from a debug work of mine, you can most probably remove these messages and ignore this.

closed 2017-08-26 by baruch