baruch/b3603 #21: Has anyone tried what the serial connection on the control board does?

danieltroger, 2020-11-10:
Is there any known protocol to control CC and CV? I bought a D3806 which has the same control board as the b3603 and hoped to be able to control it like described here: https://blog.benjames.io/2018/06/29/secret-uart-on-chinese-dcdc-converters/

but everytime I send something over the UART connection nothing comes back

rin67630, 2021-03-20:
The serial port appears to be used to program the microcontroller.

rin67630, 2021-03-20:
P.S. you might want to take a look at my active project [https://github.com/rin67630/Drok-Juntek-on-steroids](url), where I completely replaced the control unit with an ESP32 jettisoning the device in a complete new WiFi enabled realm...
;-)